fu-hsi/PMS #1: Problem with PMS and I2C

bHogan2, 2017-11-02:
Thank you for this nice library.

I have a problem where I do not get a read if certain GPIO pins are used for I2c.

If I use the default (SDA = 4 - SCL = 5) it works properly.
However if I use SDA = 0 - SCL = 2    I will not get a reads.

It's easy to test by adding ...
Wire.pins(0, 2); 
Wire.begin();

As I have hardware already using those pins I don't have a choice.
I have used SDA = 0 - SCL = 2  with many libs without a problem.
 
I didn't see anything obvious in your code. I don't understand why it doesn't work.

Appreciate any help. Thanks

fu-hsi, 2017-11-02:
Hi.
PMS sensor uses serial port, not I2C.
"Wire Library allows you to communicate with I2C / TWI devices".

You need use hardware serial or software:
https://www.arduino.cc/en/Reference/SoftwareSerial

Regards.

bHogan2, 2017-11-02:
I know that the sensor uses the serial port. I have it connected and your examples work.

It just doesn't work if you use I2C (for a display for instance)
I am running on the ESP8266. If I add 
Wire.pins(0, 2);
Wire.begin();
to your example it quits working.

If you try it on the ESP8266 you should see the problem.

fu-hsi, 2017-11-02:
OK, I'll test it tomorrow, because I do not have a complex device.

maddhin, 2017-11-03:
Do I understand you right and you connect the PMS with rx/tx pin and it stops working when you use pin 0/2 for I2C? 

I'm ceratinly not an expert on this but I once had a similar problem with an Wemos D1 mini. The 0 pin is for some startup stuff and doesn't work properly (and therefore should not be used). Took me 4ever to figure out. Best is to use the dedicated SDA/SCL pins and move the other devices to other pins - I guess having a LED or similar on pin 0 should be fine. 

You can also attach several devices to the I2C - in case this solves your issues with too many pins used.

fu-hsi, 2017-11-03:
First of all, this is not PMS library problem.
In my below example, PMS library does not included at all, and code not work with your specified code:

```
#include <Wire.h>

void setup()
{
  Serial1.begin(9600);  // GPIO2 (D4 pin on ESP-12E Development Board)
  Serial1.println("Setup");
  
  Wire.pins(0, 2);
  Wire.begin();
}

void loop()
{
  Serial1.println("I Am Alive!");
  delay(2000);
}
```

On this site:
http://arduino-esp8266.readthedocs.io/en/latest/libraries.html
they write:

> Wire library currently supports master mode up to approximately 450KHz. Before using I2C, pins for SDA and SCL need to be set by calling Wire.begin(int sda, int scl), i.e. Wire.begin(0, 2) on ESP-01, else they default to pins 4(SDA) and 5(SCL).

Pins 4 and 5 works (serial port) on my ESP-12E board.
I suggest you use what they recommends.
You must find problem elsewhere in your configuration.
Just remove my library (for sure) and test if serial port works.
If yes, my library can works too.

bHogan2, 2017-11-03:
Thanks for all your input.
My problem was solved by simply using the timeout parameter (1000ms) on the read.
It works fine with Wire.begin(0, 2). (IMO a default isn't always a recommendation.)

I'm not sure why the timeout was necessary, but it sure is nice having your lib working.
Thanks again

fu-hsi, 2017-11-03:
Strange anyway, but I'm glad it works :-)

fu-hsi, 2017-11-03:
I tested my Advanced.ino example without specifying the timeout argument.
I think this could be called a bug, but not completely.
In active mode, timeout is mandatory, but default is 0, because the default mode is passive, where timeout is not used.
read() function process only one char and back to the main program.
If u want read all, u can specify timeout or add read() function in to the loop() (not recommended, can block the program indefinitely):

```
while (pms.read(data) == false);

Serial1.println("Data:");
Serial1.print("PM 1.0 (ug/m3): ");
Serial1.println(data.PM_AE_UG_1_0);
```

In the next version I am planning a small reorganization of the function.

bHogan2, 2017-11-04:
I have a final question.
Are the pms.wakeUp() and pms.sleep() calls available in passive mode?

fu-hsi, 2017-11-04:
Yes.

fu-hsi, 2017-11-04:
To test how modes works, you can use my Windows Application:
https://github.com/fu-hsi/PMS-Tool

closed 2017-11-04 by fu-hsi