fu-hsi/PMS #2: Sleep does not work?

roger-zacharczyk, 2018-03-07:
Hello,

When calling pms.sleep() it goes to sleep but it immediately wakes up.

Is this proper behavior? What about the SET pin?

fu-hsi, 2018-03-07:
Witam.
Jesteś z Polski, więc odpowiem po polsku.
To nie jest normalne zachowanie, ale bez kodu i bez informacji o czujniku trudno cokolwiek zgadywać.
Albo coś wysyłasz po komendzie do uśpienia, albo następuje restart sensora z jakiegoś powodu.
Pinu SET nie używałem z prostego powodu. Jeśli można sterować uśpieniem komendami, to obsługa pinu SET jest zbędna, ale możesz spróbować, zamiast komendy sleep(). Jest to też jakieś wyjście.

roger-zacharczyk, 2018-03-07:
Użyłem przykładu Advanced. Zmieniłem tylko Serial na Serial1 i Serial1 na Serial.

Mam PMS7003 i Arduino Leonardo. Pomiędzy nimi na liniach UART jest konwerter poziomów logicznych.

PMS7003 podaje jakieś dane, wyglądają one sensownie.

W momencie wołania pms.sleep() na chwilę usypia (1-2s) po czym natychmiast się wybudza. Słychać jak zatrzymuje się i startuje wentylator oraz widzę na amperomierzu zmiany.

fu-hsi, 2018-03-07:
Przykład Advanced pracuje w trybie Pasywnym.
Jeśli dane przychodzą po wysłaniu komendy requestRead(), to komunikacja raczej działa poprawnie.
Niestety nie mam Leonardo, nigdy nie używałem.
Możesz jeszcze spróbować wersji Windowsowej do testów:
https://github.com/fu-hsi/PMS-Tool.

closed 2018-03-20 by fu-hsi