fu-hsi/PMS #3: pms.read returns false

bradyjoh, 2018-03-28:
Trying to use the Basic sketch just to make sure everything works, but pms.read always returns "false" so I never get to see if any data is written.

PMS7003 is connected to verified +5v and GND on pins 1/2 (labeled VCC and GND on the breakout board) and the fan spins up. Pin 9 (TX) on the PMS is connected to Serial RX on a NodeMCU/ESP8266 and RX on the PMS is connected to TX on the NodeMCU. 

Is there any troubleshooting for figuring out why a PMS sensor isn't recognized?

bradyjoh, 2018-03-28:
Interestingly, I changed the optional timeout option on pms.read() to 1000 and started getting data!

closed 2018-03-28 by bradyjoh

fu-hsi, 2018-03-28:
Unmodified basic example...?
Your problem is very similar to https://github.com/fu-hsi/PMS/issues/1.
I don't get it, but for sure, u can also check my windows application:
https://github.com/fu-hsi/PMS-Tool/releases
and confirming problem occur?

U can try also new v.1.1 branch (not released yet):
https://github.com/fu-hsi/PMS/tree/v1.1.x
If u wanna wait to parse all response from PMS, u can use:
```
readUntil(data);
```
function with default timeout set to 1000ms, but this is equal to
```
read(data, 1000);
```
in the current library.

bradyjoh, 2018-04-02:
As mentioned in the followup, it just seemed to be timing out and returning false. Increasing the timeout delay fixed the issue for me right away! 

I also don't have a tool to connect it to Windows directly, it must interface through the Arduino with the tools I have. 

Why do you recommend HardwareSerial? Is it more reliable or faster to respond? 

I'm avoiding using Hardware Serial because they're used by my NodeMCU for uploading and Serial Monitor; attaching the PMS to one of the hardware serial pins causes uploads to fail.

fu-hsi, 2018-04-02:
> Why do you recommend HardwareSerial? Is it more reliable or faster to respond?
For eliminate problems in third party code. Emulation is always just emulation.

> attaching the PMS to one of the hardware serial pins causes uploads to fail.
I known this pain, I also had to disconnect the sensor before programming, but now I use OTA update (over-the-air update) in my Air Quality Monitoring Station and problem solved.