fu-hsi/PMS #5: Null values at constant intervals

fu-hsi, 2018-11-24:
Looks familiar?
```
    id                   dt     pm1    pm25    pm10  
------  -------------------  ------  ------  --------
134853  2018-11-24 13:22:59  (NULL)  (NULL)    (NULL)
134852  2018-11-24 13:20:13      16      30        36
134851  2018-11-24 13:17:27      14      30        36
134850  2018-11-24 13:14:42      12      26        32
134849  2018-11-24 13:11:58      10      24        24
134848  2018-11-24 13:09:12       8      20        20
134847  2018-11-24 13:06:28       8      20        20
134846  2018-11-24 13:03:42       2      16        16
134845  2018-11-24 13:00:58  (NULL)  (NULL)    (NULL)
134844  2018-11-24 12:58:12      14      24        34
134843  2018-11-24 12:55:27      12      22        34
134842  2018-11-24 12:52:43      10      22        24
134841  2018-11-24 12:49:58      10      18        18
134840  2018-11-24 12:47:12       6      16        16
134839  2018-11-24 12:44:28       4      16        16
134838  2018-11-24 12:41:42       0       6         6
134837  2018-11-24 12:38:58  (NULL)  (NULL)    (NULL)
```
In passive mode I receive NULL values from PMS 7003 sensor with constant intervals.
I have not received any complaints from others, these are my observations only.

**Simple solution without modify library:**
Increase default timeout value for readUntil().

**E.g.**
```
pms.readUntil(data, 2000);
```

In next release I will use flush() after all commands to the sensor to be sure that the whole message has arrived.

YiannisBourkelis, 2019-10-02:
@fu-hsi yes I have the same issue. I am also getting NULL in passive mode. I will try increasing the timeout as you suggest and let you know.

coelner, 2020-07-08:
As noted by adafruit (https://cdn-learn.adafruit.com/downloads/pdf/pm25-air-quality-sensor.pdf , pg.13) there is a delay in the refresh rate abound 2.3 seconds max when there is no change in the particle counting, but it can speed up to 200-800ms refresh rate.

fu-hsi, 2020-07-08:
> 
> 
> As noted by adafruit (https://cdn-learn.adafruit.com/downloads/pdf/pm25-air-quality-sensor.pdf , pg.13) there is a delay in the refresh rate abound 2.3 seconds max when there is no change in the particle counting, but it can speed up to 200-800ms refresh rate.

It's referred to the Active mode (default). From official documentation:

> There are two options for digital output: passive and active. Default mode
> is active after power up. In this mode sensor would send serial data to the
> host automatically .The active mode is divided into two sub-modes: stable
> mode and fast mode. If the concentration change is small the sensor
> would run at stable mode with the real interval of 2.3s.And if the change is
> big the sensor would be changed to fast mode automatically with the
> interval of 200~800ms, the higher of the concentration, the shorter of the
> interval.

But in our example (Passive mode) response time is less than 1s, but I recommend setting to 2s.