fu-hsi/PMS #6: sleep() does not stop fan

coelner, 2018-12-05:
I use a PMS7003 and the sleep command does not stop the fan. If I short the SET pin to GND the fan stops.

fu-hsi, 2018-12-05:
I think that sleep() is not the problem, but communication with (to) PMS, and none commands works, I am right?
Just check wiring, baud rate and so on.
Depending on the module, it is not advisable to use TX/RX lines that are shared with USB (for programming).
For better understand your problem, more information is required:
- module,
- circuit.
- code.

coelner, 2018-12-05:
```
#include <SoftwareSerial.h>
#include "PMS.h"

SoftwareSerial SerialPMS;
PMS pms(SerialPMS);
PMS::DATA data;

void setup()
{
  Serial.begin(115200);   // GPIO1, GPIO3 (TX/RX pin on ESP-12E Development Board)
  SerialPMS.begin(D1,D2);  // GPIO2 (D4 pin on ESP-12E Development Board)
  pms.passiveMode();    // Switch to passive mode
}

void loop()
{
  Serial.println("Waking up, wait 30 seconds for stable readings...");
  pms.wakeUp();
  delay(30000);

  Serial.println("Send read request...");
  pms.requestRead();

  Serial.println("Wait max. 1 second for read...");
  if (pms.readUntil(data))
  {
    Serial.print("PM 1.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_1_0);

    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);

    Serial.print("PM 10.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_10_0);
  }
  else
  {
    Serial.println("No data.");
  }

  Serial.println("Going to sleep for 60 seconds.");
  pms.sleep();
  delay(60000);
}

I choosed the Advanced.ino and changed it to use Softwareserial. The PMS7003 uart module is connected via Pins D1 and D2 at a wemos D1 Mini. 
This is the used one: https://i.ebayimg.com/images/g/tr0AAOSwNDhb0t5k/s-l500.jpg

So, getting the values works fine.

fu-hsi, 2018-12-05:
You forgot to set the baud rate.

I do not think it is correct:
`SerialPMS.begin(D1,D2);  // GPIO2 (D4 pin on ESP-12E Development Board)`

begin() require speed parameter, not pin numbers:
https://github.com/arduino-org/Arduino/blob/master/hardware/arduino/samd/libraries/SoftwareSerial/SoftwareSerial.h

Read this:
https://www.arduino.cc/en/Reference/SoftwareSerialBegin

And set proper baud rate to 9600 (PMS::BAUD_RATE).

coelner, 2018-12-05:
Sorry, I fixed the sketch. ( I removed sketch and library to test others. Therefore I had to rewrite the sketch)

fu-hsi, 2018-12-05:
If still does't work, I recommend use another library which implement Serial functionality, like:

- AltSoftSerial,
- NeoSWSerial.

coelner, 2018-12-05:
Ok, I have no time to test those libraries. Instead I use the original sketch with hardware serial. Even this does not stop the fan. The measuring and reporting works, at least I get values.

coelner, 2018-12-30:
Both libraries doesn't work with the esp8266. 
I tried again the original sketch which works. I'm currently on the latest trunk. 
SoftwareSerial does not work, but the sleep command is send.

fu-hsi, 2018-12-30:
After:
`pms.sleep();`
You can add:
`SerialPMS.flush();`

If it doesn't help, you can always use dedicated PMS pin:

> SET - Standby function, when pulled down the module does not send information over serial and goes into a low power state.

coelner, 2018-12-30:
I could use those separate pins but it should be possible even with the softwareserial library. It is not very special at all, therefore must be somehow a solution.

I sniffed the traffic:
```
ssterm -b 9600 -o split /dev/ttyUSB0
42 4d e4 00 01 01 74 42  4d e2 00 00 01 71 40 4d  |BM....tBM....q@M|
e4 00 00 01 73 42 4d e4  00 01 01 74 42 9a e2 00  |....sBM....tB...|
00 01 71 40 4d e4 00 00  01 73                    |..q@M....s      |
```
 * wakeUp
 * requestRead
 * Sleep [but wrong leading '40']
 * WakeUp
 * Garbage [42 9a e2 00 00 01 71]
 * Sleep [but wrong leading '40']

coelner, 2018-12-30:
I tried the espsoftwareserial
```
$ssterm -b 9600 -o split --rx-nl crlf /dev/ttyUSB0
42 4d e1 00 00 01 70 42  4d e4 00 01 01 74 42 4d  |BM....pBM....tBM|
e2 00 00 01 71 42 4d e4  00 00 01 73 00 00 42 4d  |....qBM....s..BM|
e1 00 00 01 70 42 4d e4  00 01 01 74 42 4d e2 00  |....pBM....tBM..|
00 01 71 42 4d e4 00 00  01 73 
```
 * passiveMode
 * WakeUp
 * requestRead
 * sleep
 * 00 00
 * passiveMode
 * wakeUp
 * requestRead
 * sleep
It works with the sleep.  Do you fill a issue at esp8266 git?

fu-hsi, 2018-12-30:
So now everything works? That's good.

I don't remember If I tested library with software serial implementation.
I use Hardware Serial always when possible.

My Station is online now and I can't modify:
https://github.com/fu-hsi/AQMS

As I suggested, read this: #5 and replace all instances:
```
_stream->write(command, sizeof(command));
```
with
```
_stream->write(command, sizeof(command));
_stream->flush();
```
just in case. This command wait for transmit all bytes.

coelner, 2018-12-31:
I will test it. But somehow the commands were changed if I used the native library.
I close the issue because it is not in your library.

closed 2018-12-31 by coelner