fu-hsi/PMS #8: No particles with diameter

chappy1978, 2019-01-13:
So I am building an Air Quality Meter and I wanted to use your library. Because it is the easiest to read and use. The only problem is it doesn't have any of the "particles with diameter beyond 0.x um in 0.1L of air." They are the last three data packets in the PMS's UART package. I'll will try to add them in my own branch but Im not a big C++ guy.
   Jeff

coelner, 2019-01-16:
It could be separated because  the length field get checked. The 3003 differs from 5003/7003 from the length (because of missing values for 0.1L).
What if the constructor creates the bare minimum. If the user calls the constructor with a parameter the extended values get accessible.

fu-hsi, 2019-01-16:
It's on todo list.

closed 2019-01-16 by fu-hsi

mano3m, 2021-08-16:
@fu-hsi would you be interested in a pull request that handles this?