fu-hsi/PMS #10: Arduino compatibility

Roberto6969, 2019-01-29:
Could be this library adopted to Ardino (Uno, Mega, ...)?

Thank You!

fu-hsi, 2019-01-29:
Hi.
What does not work?
Paste error.

Roberto6969, 2019-01-29:
It seems as speed is not setup correctly (but it is, 9600 in this case). This is from serial monitor: 
[image]

fu-hsi, 2019-01-29:
Yes, but without circuit I don't know what pins you are using.
I don't have this board and I see that many UART ports is available.
Just check wiring, try another pins, software or hardware serial.
Most of the problems comes from bad connection.

Sorry.

Roberto6969, 2019-01-29:
2560 has its own RX/TX pins (RX=4, PE0, TX=5, PE1). I believe this is the problem.
Do you mean pin definition in your code? I can not find any pin definition. From the HW point of view is everything OK because board/PMS works with other library

fu-hsi, 2019-01-29:
> I can not find any pin definition.

You are right, my mistake. I mean Serial, Serial1, Serial2...
But when you use SoftwareSerial, you can choose any pins.

Roberto6969, 2019-01-29:
Yes, you are right - I have to use SoftwareSerial also. In my case definition is following:
// Using AltSoftSerial:
//   Pin 4: Digital pin 9 (there is no choice)
//   Pin 5: Digital pin 8 (there is no choice)
//   in file pmsConfig.h: Enable #define PMS_ALTSOFTSERIAL
//   Install DrDiettrich' fork of AltSoftSerial Library: https://github.com/DrDiettrich/AltSoftSerial.git

TwinedGemini, 2019-02-11:
Hi @fu-hsi I'm in the same boat as roberto, although I might be a newbee at this (pls keep in mind). I'm trying to make your code work with SoftwareSerial but can't seem to make it work. Any guidance on the matter? 

As of now i've tried to define softserials as pin (3,2). 
` #include <SoftwareSerial.h>
#include "PMS.h"

SoftwareSerial pmsensor(3,2);
PMS pms(Serial);
PMS::DATA data;`

adrianogf, 2019-06-18:
> Hi @fu-hsi I'm in the same boat as roberto, although I might be a newbee at this (pls keep in mind). I'm trying to make your code work with SoftwareSerial but can't seem to make it work. Any guidance on the matter?
> 
> As of now i've tried to define softserials as pin (3,2). my code below:
> 
> #include <SoftwareSerial.h>
> #include "PMS.h"
> 
> SoftwareSerial pmsensor(3,2);
> PMS pms(pmsensor);
> PMS::DATA data;

Hi.

I have same issue. How to connect to Arduino using SoftwareSerial??

Thank You.

Roberto6969, 2019-06-18:
Try to avoid using Software emulation of serial port if you want to live in peace for long time ;-)
This PMS library works great so the best way is to use board which has two HW ports. The same is valid for other PMS libraries!

arduiNope, 2019-07-04:
Hi all. I am also relatively new and I have been trying to set up using this. I have the library loaded and am using the basic and advanced example but they are returning "no data". I have the pms1003 set up on an uno with the gnd, vcc and TX&RX pins (1,0). I also have a logic converter for the tx/rx pins. I'll attach a diagram.   I was wondering if anyone has any insight into why this might not be working. Do I need to alter something on the code that I am missing? Thanks in advance for any help!

[
[image]
](url)

seeya, 2019-08-23:
I'm using a `WEMOS D1 R2` with SoftwareSerial. Seems to be working fine for me.

The TX and RX pins of the `pms5003` are connected to my arduino at `TX->D3` and `RX->D4`.

Below is the code that I'm using. Hope it helps
```
#include "PMS.h"
#include <SoftwareSerial.h>

SoftwareSerial pmsSerial(D3, D4);

PMS pms(pmsSerial);
PMS::DATA data;

void setup()
{
  Serial.begin(115200);   
  pmsSerial.begin(9600);  
}

void loop()
{
  if (pms.read(data))
  {
    Serial.print("PM 1.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_1_0);

    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);

    Serial.print("PM 10.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_10_0);

    Serial.println();
  }

  // Do other stuff...
}
```

CAPlevel, 2019-12-22:
I'm trying to use seeya post to solve my issue I'm migrating from another library to https://github.com/fu-hsi  Library but I get no response, I need to use D7 & D8 (also try with GPIO  13,15) in a wemos and it compiles ok but does not work. here is my code:

---------------------------------------------------------------------------------------------------
#include "PMS.h"

#include <SoftwareSerial.h>
SoftwareSerial pmsSerial(13,15); // RX | TX / D7 = GPIO 13     , GD8 = GPIO 15

PMS pms(pmsSerial);
PMS::DATA data;

void setup()
{
  Serial.begin(9600);   // GPIO1, GPIO3 (TX/RX pin on ESP-12E Development Board)
  pmsSerial.begin(9600);  // D7 = GPIO 13     , GD8 = GPIO 15
  Serial.print("started...");
}

void loop()
{
  if (pms.read(data))
  {
    Serial.print("PM 1.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_1_0);

    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);

    Serial.print("PM 10.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_10_0);

    Serial.println();
  }

  // Do other stuff...
}

ismailanil, 2021-02-19:
> I'm using a `WEMOS D1 R2` with SoftwareSerial. Seems to be working fine for me.
>
> The TX and RX pins of the `pms5003` are connected to my arduino at `TX->D3` and `RX->D4`.
>
> Below is the code that I'm [...]

Hello
just small update from my side

I searched and tried many codes from various sources but ended up with no sensor data even the code provided by @fu-shi did not worked for arduino uno or mega boards

This revised code worked well for me on uno, mega, and seeeduino Lotus boards for both PMS5003 and PMS7003 sensors

Thank you very much Dear @fu-shi and @seeya

A small correction:
Just delete letter 'D' from "pmsSerial(D3, D4)" and type only digital pin numbers as: SoftwareSerial pmsSerial(3, 4);

For example, if you connect TX cable to D4 and RX cable to D5, code part becomes: SoftwareSerial pmsSerial(4, 5);

Million thanks....

gmrock, 2023-09-17:
Anybody has luck with raspberry picow?