fu-hsi/PMS #11: is PM 0.5 and 0.3 integrated??

TwinedGemini, 2019-02-11:
Hi @fu-hsi 

is it possible to integrate a functionality for capturing sensing data on Particles of 0.5 um and 0.3 um in you're library? as the PMS7003 can capture these.
maybe you already have this functionality, I've just not been able to find it in your doc.

best regard

fu-hsi, 2019-02-11:
PMS7003 captures:

- PM1.0,
- PM2.5,
- PM10.

For measuring air quality you only need PM10 and PM2.5 in some countries.

closed 2019-02-14 by fu-hsi

MattMills, 2021-09-01:
Per this doc: https://www.aqmd.gov/docs/default-source/aq-spec/resources-page/plantower-pms5003-manual_v2-3.pdf

The last two digits of the model number are the min distinguishable particle diameter, 03 = 0.3 micrometer, 05=0.5 micrometer, 10 = 1.0 micrometer, 25=2.5 micrometer.