fu-hsi/PMS #12: Only zero values on ESP32 Wroom

CaNiRaspUsomethinG, 2019-03-01:
Hi,

with ESP32 Wroom Dev Board I get only 0 values for all PM sizes every second.
I'm powering the PMS 7003 with 5V (with 3.3V on RX TX)

If I put this in loop: Serial.println(Ser1_33_32_PMS7003.read());

I get every second changing, but not plausible measurement values. See output example:

117:50:38.875 -> 203
17:50:39.466 -> 77
17:50:39.504 -> 0
17:50:39.537 -> 50
17:50:39.605 -> 0
17:50:39.639 -> 0
17:50:39.712 -> 0
17:50:39.748 -> 0
17:50:39.783 -> 205
17:50:40.290 -> 77
17:50:40.358 -> 0
17:50:40.396 -> 50
17:50:40.429 -> 0
17:50:40.497 -> 0
17:50:40.534 -> 0
17:50:40.602 -> 0
17:50:40.635 -> 205
17:50:41.230 -> 77
17:50:41.264 -> 0
17:50:41.297 -> 32
17:50:41.367 -> 0
17:50:41.401 -> 0
17:50:41.469 -> 0
17:50:41.503 -> 0
17:50:41.571 -> 183

Thank you for your help!

fu-hsi, 2019-03-01:
And where is your full code?

CaNiRaspUsomethinG, 2019-03-01:
Hi and sorry, here it is:

#####

// lib for using hardwareserials with ESP32
#include <HardwareSerial.h>
HardwareSerial Ser1_33_32_PMS7003(1);

// PMS OPC
#include "PMS.h"
PMS pms(Ser1_33_32_PMS7003);
PMS::DATA data;

void setup() {

  /***************************************************
    SERIAL PORTS CONFIGURATION - BEGIN
  ****************************************************/
    Serial.begin(9600);
  // let serial console settle
  delay(10);

  Ser1_33_32_PMS7003.begin(9600, SERIAL_8N1, 33, 32);
  delay(10);
  /***************************************************
    SERIAL PORTS CONFIGURATION - END
  ****************************************************/
} // setup end

void loop() {

// next for lines: just checking if serial data is coming from sensor
  //if (Ser1_33_32_PMS7003.read() > 0) {
  //delay(50);
  //Serial.println(Ser1_33_32_PMS7003.read());
  //}

// the active example function
 if (pms.read(data))
  {
    Serial.print("PM 1.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_1_0);

    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);

    Serial.print("PM 10.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_10_0);
}}

fu-hsi, 2019-03-01:
`Serial.println(Ser1_33_32_PMS7003.read());`
I don't get it, what you want to achieve with this code.
It is totally wrong. read() just returns boolean.
You did not give any output log with above code, so I don't know what is printed out now.
It is worked with ESP32 for sure, so check wiring again and your configuration (pins).
Just use attached examples and connect properly without changing in the code.
If everything work, then start modify your code 😃 
Sorry, I don't' have PMS on my desk anymore.

CaNiRaspUsomethinG, 2019-03-01:
Thank you for your response.

The lib works like it should on my ESP32! The sensor just needed several minutes to start showing the bigger values!

With the Serialprintln I wanted to pass the pmsSerial input to serial like this:

 if (pmsSerial.available()) {
    Serial.write(pmsSerial.read());
  }

Anyway, it works now, thank you!

closed 2019-03-01 by CaNiRaspUsomethinG