fu-hsi/PMS #13: ESP8266 NodeMCU CP2102

vicolodo, 2019-03-28:
Hello @fu-hsi ,
i would like to use the PMS library in a .ino sketch for the board NodeMCU in subject.
I need to understand which PMS5003 pins must be connected to which NodeMCU pins and how to specify in the .ino which NodeMCU pins i am using.
I saw somewhere that +5V can be got from NodeMCU VIN ..is it true?
I am going to connect in this way:
NODEMCU -- PMS5003:
VIN -- PIN1
GND -- PIN2
RX -- PIN5 (TXD)
TX -- PIN4 (RXD)
Is it right?
I have to connect the PMS5003 PIN3 (SET) and PIN6 (RESET) too, or they can be left unconnected?

THANK YOU!

fu-hsi, 2019-03-28:
Hi.

Yes, it looks fine.
You can leave SET and RESET pins floated.

Regards.

kalapontsai, 2019-03-29:
> I saw somewhere that +5V can be got from NodeMCU VIN ..is it true?

I do use NodeMCU Vin for PSM5005 V+.
need to make sure that the power supply is enough, 5V 500ma could be fail to start PMS, you will find the PMS fan is not working.

fu-hsi, 2019-03-29:
If you are supply NodeMCU from USB then VIN can have 5V.
You can't broke anything, just connect and check.
VIN is Voltage Input for powering module, but it can proivide voltage if you power supply from USB socket.
Ther is not diode in circuit for CH340G.

https://circuits4you.com/wp-content/uploads/2017/12/nodemcu-circuit-diagram.png

vicolodo, 2019-03-30:
Thank you @fu-hsi, @kalapontsai .
In my case, i have CP2102 instead of CH340G, i hope there is no diode also in this circuit.

I have two other questions:
I know that in NodeMCU board, the Serial pins TXD0 and RXD0 are connected to USB .
Do you think there is conflict between micro USB adapter and PMS5003?
To avoid conflict, is it enough to leave the PMS5003 unconnected while the USB is connected to my pc?

And sorry for these questions, i am a newbie:
after uploading the sketch (for example, Basic.ino), the USB can still work when i connect it to my pc and i want to upload other sketches?
Perhaps do i have to press the Flash button to put the board in Flash mode?
And finally: i see in the Basic.ino sketch that 'normal' messages are sent to TXD1 (via Serial1.println).
How can i see these messages?

In my case i am going to use your lib with Blynk lib, and so the pms data is sent through wifi to my smartphone Blynk app, so actually i dont need the Serial1 prints, but they could be useful for debug purposes...
Let me know, thank you again!!

fu-hsi, 2019-03-30:
> Do  you think there is conflict between micro USB adapter and PMS5003?

Yes. That's whay You can't use USB for programming when PMS is connected.
To get around this I use OTA (over-the-Air programming).
You can also use any other pins for TX/RX with software serial library to avoid this conflict.

> after  uploading the sketch (for example, Basic.ino), the USB can still work when i connect it to my pc and i want to upload other sketches?

No. See the first answer.

> Perhaps do i have to press the Flash button to put the board in Flash mode?

I never did use this, so I don't know.

> And  finally: i see in the Basic.ino sketch that 'normal' messages are sent to TXD1 (via Serial1.println).
How can i see these messages?

You muse connect external CP2102 or CH340G module to TXD1 and then to PC or you can use HC-06 and your Smartphone for receive messages (https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en).

[image]

Regards.

vicolodo, 2019-03-30:
thank you @fu-hsi 
yes perhaps i have to buy one of such cp2102 , ch340g , hc06 modules
but in the meantime i would try to use software serial...about this i have a question:
to use software serial are needed big changes in your lib?
i forked the lib to my user, so i could try to add the changes there...

fu-hsi, 2019-03-30:
> to use software serial are needed big changes in your lib?

Nope. Small changes is required.
For example: #14.

PMS object requires Stream object as argument.
You can pass hardware or software serial object.

vicolodo, 2019-03-30:
ah thank you, @fu-tsi, seems simple, i will try.
but do you think could be better to use hardware serial instead of software serial?

fu-hsi, 2019-03-30:
I use hardware serial when possible.
Instead of SoftwareSerial you can use better implementations like:

- AltSoftSerial,
- NeoSWSerial.

Roberto6969, 2019-03-31:
The best way to avoid many problems is to use hw serial port. Nowadays there are many microcontrollers/developer boards which has at least two HW  serial ports. In my case I using MEGA2560 Core (or Pro) and with this library works great

vicolodo, 2019-03-31:
Ok thank you for the suggestions.
I tried the sensor using pms and 'standard' software serial lib, using GPIO04 and GPIO05 pins.
At first time it did not work, i had only 'No Data' readings.
Then i found information that ESP8266 does not work with 'standard' software serial, but needs a specific library 'espserial' .
So i installed it , uploaded the sketch #14 , and it worked.
I have a question, now, about the readings:
the values are not stable:
for example, in my home, i get for pm2.5 in sequence :
18 µg/m³
10 µg/m³
6 µg/m³
11 µg/m³
15 µg/m³
...
so i wonder if the sensor is working well, or it gives random results

perhaps i have to calculate the medium value of the readings?

thank you!

vicolodo, 2019-04-02:
hello @fu-tsi, i tried to use default mode ( basic.ino logic) , and the readings now have changed, and are more stable
now i read
pm1.0 10 
pm2.5  16
pm10 22
and these values change only a little in the time (+- 1 or 2 points)
seems these values are correct and instead the old values in passive mode are not correct...

closed 2019-04-04 by fu-hsi