fu-hsi/PMS #15: avrdude: stk500_recv (): programmer is not responding

leandronaya, 2019-05-12:
Hi,

I am trying to use your code, with a Geniunie Arduino Uno, connected with de AirQuality Monitor (from Plantower) with de PM2.5 Sensor Adpater with a IO Expansion Shield V7.1. Just as the wiki indicates to connect the Air Quality Sensor to the arduino.

When I´m trying to upload the code to the Arduino it generates the following error code:

avrdude: stk500_recv (): programmer is not responding
avrdude: stk500_getsync () attempt 1 of 10: not in sync: resp = 0xb0
avrdude: stk500_recv (): programmer is not responding
avrdude: stk500_getsync () attempt 2 of 10: not in sync: resp = 0xb0
avrdude: stk500_recv (): programmer is not responding
avrdude: stk500_getsync () attempt 3 of 10: not in sync: resp = 0xb0
avrdude: stk500_recv (): programmer is not responding
avrdude: stk500_getsync () attempt 4 of 10: not in sync: resp = 0xb0
avrdude: stk500_recv (): programmer is not responding
avrdude: stk500_getsync () attempt 5 of 10: not in sync: resp = 0xb0
avrdude: stk500_recv (): programmer is not responding
avrdude: stk500_getsync () attempt 6 of 10: not in sync: resp = 0xb0
avrdude: stk500_recv (): programmer is not responding
avrdude: stk500_getsync () attempt 7 of 10: not in sync: resp = 0xb0
avrdude: stk500_recv (): programmer is not responding
avrdude: stk500_getsync () attempt 8 of 10: not in sync: resp = 0xb0
avrdude: stk500_recv (): programmer is not responding
avrdude: stk500_getsync () attempt 9 of 10: not in sync: resp = 0xb0
avrdude: stk500_recv (): programmer is not responding
avrdude: stk500_getsync () attempt 10 of 10: not in sync: resp = 0xb0

I do not know if this is a known problem or if you know how to fix it.
As a note, if I disconnect the PM2.5 Sesor adapter from the Expansion Shield it is possible to load the code, but it does not work when I connect it again. Finally, I have an MQ9 sensor that I´m using with the IO expansion shield and it works properly. So, I´m thinking the problem may be in the code for the Air Quality Sensor ( I have two sensors, just in case, and both have the same problem, and I try with mac and pc). 
I will be very grateful for your help, greetings from now,

fu-hsi, 2019-05-16:
Hi.
If USB shares the same data lines with Serial, then yes, this is normal...
Just disconnect PMS module, program and connect again.
Please read the closed issues.
I answered what i recommend for this situations.
Use software serial with another free pins or OTA update if possible.
Full example of my project with OTA:
https://github.com/fu-hsi/AQMS
This is not a library problem and I do not have your Shield, so I can not help more.

leandronaya, 2019-05-17:
> Hi.
> If USB shares the same data lines with Serial, then yes, this is normal...
> Just disconnect PMS module, program and connect again.
> Please read the closed issues.
> I answered what i recommend for [...]
Thanks fu-hsi. I solved it in the way you indicate. Another point that I saw is that it uses the virtual Rx and Tx ports it is possible to program it directly.

closed 2019-05-17 by leandronaya

BunnyShoppers, 2019-09-01:
It can happen because of these two problems.
1. Maybe you are trying to upload program while your rx and tx pins are connected with other components.
2. Or maybe your arduino is bricked.
For 1st problem solution is just disconnect wires rx,tx and try to upload the sketch. 
For 2nd problem use another arduino to upload program on it(SPI connection). 
For 2nd problem i found this article helpful: [Programmer is not responding Arduino](https://www.sheebot.com/2019/07/sketch-not-uploading-through-usb.html)

sailkiran, 2019-10-28:
> It can happen because of these two problems.
>
> 1. Maybe you are trying to upload program while your rx and tx pins are connected with other components.
> 2. Or maybe your arduino is bricked.
>    For 1st [...]

hello bynnyshoppers,

my arduino is bricked , your solution worked for me. thanks for that.
i've question, i'm using a sensor where i will get the values from RX to the board and i need to sd card module parallelly, in this case, I cannot remove the rx because my output is based on that, 
 
any solution please reply at the earliest.