fu-hsi/PMS #16: brak odczytów po restarcie PMS

sotb11, 2019-06-11:
Robię update programu przez webupdate. Dotąd wykonałem dwa takie update w momencie, gdy PMS3003 jest uśpiony. Po restarcie ESP i wywołaniu pms.wakeUp() nie są dokonywane pomiary. Oczywiście wentylatora nie słychać. Dopiero po przejściu całej pętli i ponownym pms.wakeUp() pomiary są wykonywane normalnie. Czytałem w pliku expert komentarz, że trzeba pms wybudzić ręcznie, bo po restarcie może nie dokonać pomiaru. 
Podobna sytuacja jest, gdy wcisnę reset na płytce. Też pms nie jest wybudzany przy pierwszym przejściu pętli. 
Czyżbym musiał pms uśpić w setup() i zaraz go budzić?

fu-hsi, 2019-06-12:
Witam.
Raczej nie pisałem o ręcznym wybudzaniu, problem może być w pierwszej pętli,
ale to jest spowodowane raczej tym, że ESP wysyła wtedy na serial różne śmieci (i/lub informacje debugujące po flashowaniu lub resecie). Te dane w połączeniu z komenda trafiają potem do PMS, ktory nie jest w stanie poprawnie ich zinterpretować, chociaż wg mnie powinien, bo są sumy kontrolne i to co jest przed i po poleceniem nie powinno mieć znaczenia, jednak ja to też zauważyłem, dlatego o tym wspomniałem w komentarzu.
Ja bym proponował użyć innych, alternatywnych pinów, aby nie używać seriala, który jest współdzielony z USB, np. przez swap() jak robię to tutaj:
https://github.com/fu-hsi/FuGPS/blob/master/examples/FuGPS_AutoBaud/FuGPS_AutoBaud.ino.
Można też poszukać jakiejś biblioteki do softwerowego seriala na ESP.
Innych pomysłów nie mam i o szczegółach ciężko teraz mówić z perspektywy czasu.
Aktualizacje robi się tak rzadko, że utrata jednego odczytu nie jest dla mnie problemem, a stacja działa od samego początku na tym kodzie:
https://github.com/fu-hsi/AQMS.

Pozdrawiam.

sotb11, 2019-06-12:
Oczywiście. Rozumiem, że nie ma takiej opcji żeby go "ręcznie" obudzić :)
Przejrzałem Twój skrypt i zauważyłem, że PMS jest wybudzany u Ciebie również w setup(). 
Po wprowadzeniu tej modyfikacji u mnie w setupie problem się rozwiązał.
Dziękuję za naprowadzenie i również pozdrawiam.

closed 2019-06-12 by sotb11