fu-hsi/PMS #20: Unable to see readings w Arduino UNO

jsy360, 2020-04-30:
Hello,

Thanks for this library. 

I am attempting to use the PMS5003 via breakout board with an Arduino UNO. 

I am using SoftwareSerial.h to create a Serial1 with Uno Pins 2,3.  

However, I cannot see any readings from the PMS5003 unit. 

Please advise on my next step? Thank you

Jason

My setup: 

Sensor TX -> Arduino PIN 2
Sensor RX -> Arduino PIN 3
5V to 5V, GND to GND
Reset and EN pin not plugged in

[image]

My code:
```

#include <PMS.h>
#include <SoftwareSerial.h>

SoftwareSerial Serial1(2,3);

PMS pms(Serial);
PMS::DATA data;

void setup()
{
  Serial.begin(9600);   // GPIO1, GPIO3 (TX/RX pin on ESP-12E Development Board)
  Serial1.begin(9600);  // GPIO2 (D4 pin on ESP-12E Development Board)
}

void loop() {
  
  if (pms.read(data))  {
    
    Serial.print("reading data");
    Serial1.print("PM 1.0 (ug/m3): ");
    Serial1.println(data.PM_AE_UG_1_0);

    Serial1.print("PM 2.5 (ug/m3): ");
    Serial1.println(data.PM_AE_UG_2_5);

    Serial1.print("PM 10.0 (ug/m3): ");
    Serial1.println(data.PM_AE_UG_10_0);

    Serial1.println();
  }
  // Do other stuff...
}
```

munzc, 2020-05-01:
> My setup:
> 
> Sensor TX -> Arduino PIN 2
> Sensor RX -> Arduino PIN 3
> 5V to 5V, GND to GND
> Reset and EN pin not plugged in

I was having this same exact issue with receiving no data.  I am not sure why this does not work, but I was able to get data readings by referencing seeya's post on this thread.  https://github.com/fu-hsi/PMS/issues/10

Use this code with the Serial monitor at 115200 baud (you should not have to modify your wiring):

```
#include "PMS.h"
#include <SoftwareSerial.h>

SoftwareSerial pmsSerial(2, 3);

PMS pms(pmsSerial);
PMS::DATA data;

void setup()
{
  Serial.begin(115200);   
  pmsSerial.begin(9600);  
}

void loop()
{
  if (pms.read(data))
  {
    Serial.print("PM 1.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_1_0);

    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);

    Serial.print("PM 10.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_10_0);

    Serial.println();
  }

  // Do other stuff...
}
```

fu-hsi, 2020-05-01:
If the output port is slow, the input port will get a hiccup.
That's why @munzc may be right.
I also did this with the FuGPS library, which is why I do not recommend using SoftwareSerial, it is the worst library for serial port support.
Always use a hardware port when possible or other library.

Quote from my other library:

> Please, don't use SoftwareSerial if you must capture more than one type of message, it is very ineffective.
> You can experiment with setting proper input and output baud rate, but the best solution to me is to use better implementations of Serial:
> 
> - HardwareSerial (always the best way).
> - NeoSWSerial (ESP8266 not supported).
> - AltSoftSerial (pins 8 & 9 for a Nano, ESP8266 not supported).

jsy360, 2020-05-01:
Hey fu-hsi: so I tried this without Software Serial, but still no results. Using the <NeoSWSerial.h> library, that I picked up from your suggestion to use the FuGPS. (I took out the FuGPS because I figured it wasn't needed).

I even tried to Level Change from 3.3V to 5V for the Breakout Board's TX and RX, as a possible solution, but still no results. 

Ever reach a point where you think something will never get solved? :S 

Current code:

```
#include <PMS.h>
#include <NeoSWSerial.h>

NeoSWSerial in(2,3);// RX, TX

PMS pms(in);
PMS::DATA data;

void setup()
{
  Serial.begin(115200);   
  in.begin(9600); 
}

void loop()
{
  if (pms.read(data))
  {
    Serial.print("PM 1.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_1_0);

    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);

    Serial.print("PM 10.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_10_0);

    Serial.println();
  }

  // Do other stuff...
}
```

munzc, 2020-05-01:
I did some research and I found a way to use the serial hardware port (digital pin 0 and 1 on the Uno) instead of using SoftwareSerial as recommended by @fu-hsi.  

You must upload this code to the Uno _before_  plugging any sensor wires into the Arduino RX or TX ports (D0 & D1 on the Uno), otherwise you'll have an error uploading.  I believe this is because the Arduino uses D0 and D1 to communicate with the PC, so plugging other wires in there will interfere with the serial connectivity when initially uploading code.

Then wire as follows:
Sensor TX -> Arduino DX (pin D0)
Sensor RX -> Arduino TX (pin D1)
Sensor GND -> Arduino digital GND
Sensor VCC -> Arduino 5V

```
#include "PMS.h"

PMS pms(Serial);
PMS::DATA data;

void setup()
{
  Serial.begin(9600);   
  pms.activeMode();  
}

void loop()
{
  if (pms.read(data))
  {
    Serial.print("PM 1.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_1_0);

    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);

    Serial.print("PM 10.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_10_0);

    Serial.println();
  }

  // Do other stuff...
}
```
Using `PMS pms(Serial)` automatically references D0 and D1 as RX and TX ports, so you don't have to manually write in the pins as you would with SoftwareSerial.  Also, I found that adding `pms.activeMode()` into `setup()` is helpful as it makes sure that the sensor is at default mode.

Here is an example of the output. @jsy360 Please let me know if you try this out and it works for you! I'm  interested to see if you get similar outputs as mine under normal indoor conditions.
[image]

fu-hsi, 2020-05-01:
@munzc this is true even in ESP. Always disconnect the sensor when programming, unless using OTA :)
This is true even in ESP. Always disconnect the sensor when programming, unless using OTA :)
I could never understand why people use SoftwareSerial instead of the built-in serial port :)

fu-hsi, 2020-05-01:
@jsy360 I have UNO, but no extra PMS to figure it out, where is the problem :(
I would have to disassemble my only weather station that has been operating for years (powietrze.lomza.pl), but I don't want to break it :)

jsy360, 2020-05-01:
hey @munzc sorry but it did not work. 

I uploaded your code to the UNO without anything plugged in first - but no result. Followed your wiring as well.

Maybe my PMS is malfunctioning(?). It's definitely on - when I put my ear put to it, I hear a humming sound. 

The results that I see from your suggested code:

[image]

fu-hsi, 2020-05-01:
> ```
> #include "PMS.h"
>
> PMS pms(Serial);
> PMS::DATA data;
>
> void setup()
> {
>   Serial.begin(9600);
>   pms.activeMode();
> }
>
> void loop()
> {
>   if (pms.read(data))
>   {
>     Serial.print("PM 1.0 (ug/m3): [...]

This code is wrong. You use the same Serial port for writing and reading.
@jsy360 , better start from beginning and use code examples.

jsy360, 2020-05-01:
@fu-hsi just a general question:

when you write 

`Serial1.print("PM 1.0 (ug/m3): ");`

does that print out on the serial monitor? does that mean Serial.print and Serial1.print both print on to the same monitor?

fu-hsi, 2020-05-01:
https://www.arduino.cc/reference/en/language/functions/communication/serial/
UNO has only one Serial, not Serial1.
So, for PMS I recommend using Serial.
For debug... you must use another Serial port (which doesn't exist), so you must use SoftwareSerial.
But for display, you must have additional usb rs232 converted to display data on your PC ;)
You can use HC-05/06 Bluetooth Module for that.
I known, we complicated a lot, but you can send data everywhere you need.

munzc, 2020-05-02:
Sorry about that post before, I didn't realize that I was using the hardware serial port for both writing and reading (I'm still a beginner with this).  

In hindsight, the best thing for this is to probably use an Arduino Due or Mega that has multiple serial ports, or using a HC-05/06 module so you can use the hardware port for the PMS.  

[https://arduino.stackexchange.com/questions/34126/whats-the-difference-between-all-the-software-serial-libraries-which-one-is-ar](url) 
I read here that the next best thing to a hardware serial port is `AltSoftSerial`.   I have this code on my UNO with (Sensor TX -> Pin 8) and (Sensor RX -> Pin 9). 
```
#include "PMS.h"
#include <AltSoftSerial.h>

AltSoftSerial softSerial; //(Pin 8: RX), (Pin 9: TX)

PMS pms(softSerial);
PMS::DATA data;

void setup()
{
  Serial.begin(9600);   // Hardware serial port: prints data
  softSerial.begin(9600);  // Software serial port: communicates w/ PMS5003 sensor via pms() library
}

void loop()
{
  if (pms.read(data)){
    
    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);
    Serial.println();
    delay(1000);
  }
}
```
[image]

jsy360, 2020-05-04:
Hey @fu-hsi @munzc thanks for the efforts.

So I actually found a Due, so I am using its pin 18, 19 for its Serial1. 

Still nothing, but I just wanted to check with you guys if all the code and wiring seemed correct. Please let me know? 

At this point, I might just conclude that the PMS5003 unit itself might be broken(?) and that I should order another one.

Code:
```

//#include <SoftwareSerial.h>
#include <PMS.h>

//SoftwareSerial Serial1;// RX, TX

PMS pms(Serial1);
PMS::DATA data;

void setup()
{
  Serial.begin(115200);   
  Serial1.begin(9600); 
}

void loop()
{
  if (pms.read(data)){
    
    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);
    Serial.println();
    delay(1000);
  }
}
```
Note: I commented out SoftwareSerial since it is my understanding that Serial1 in built-in and does not need declaring, like Serial

Wiring:
Sensor TX -> pin 19 (RX1)
Sensor RX -> pin 18 (TX1)
5V-5V; GND-GND
[image]

fu-hsi, 2020-05-04:
Looks good for me. Fan working? You can also add this lines after PMS initialization in setup() function:
```
pms.activeMode();
pms.wakeUp();
```
Before buying a new module you can try my Windows application first:
https://github.com/fu-hsi/PMS-Tool

fu-hsi, 2020-05-04:
You don't need to have my library to check if the module is sending something.
Just try to see what's happening on the port without my library.

jsy360, 2020-05-04:
> Looks good for me. Fan working? You can also add this lines after PMS initialization in setup() function:
> 
> ```
> pms.activeMode();
> pms.wakeUp();
> ```
> 
> Before buying a new module you can try my Windows application first:
> https://github.com/fu-hsi/PMS-Tool

-Fan is working. I can the whir when I put my ear up to it. I added the lines but still nothing. This is very confusing.

-Would love to try your windows app but I am on a mac.

> You don't need to have my library to check if the module is sending something.
> Just try to see what's happening on the port without my library, eg. Putty, PMS-Tool.

I'm a little confused by this: how do I check to see if the module is sending something with your library? PMS Tool is your windows app, and Putty seems like some kind of virtual Serial Monitor?

fu-hsi, 2020-05-04:
For MAC users:
https://apple.stackexchange.com/questions/32834/is-there-an-os-x-terminal-program-that-can-access-serial-ports

munzc, 2020-05-04:
One thing to keep in mind is bad wire connections.  Yesterday I was having a problem with getting no readings and it turned out the pin on one of my wires came loose, so I just had to swap it out for a new one and that fixed it.  So maybe double check the sensor's RX and TX wires with an ohmeter or swap them. 

Also, (this may be obvious, but I didn't know this) did you go into Tools>Manage Libraries and install the PMS Library?  The fan would still run even without this if you supply power to it.

You can also try this code out and see if it works.  This helped me with troubleshooting.
https://learn.adafruit.com/pm25-air-quality-sensor/arduino-code

jsy360, 2020-05-04:
> One thing to keep in mind is bad wire connections. Yesterday I was having a problem with getting no readings and it turned out the pin on one of my wires came loose, so I just had to swap it out for [...]

Hey @munzc - yeah I swapped out the wires, and yes the PMS library is installed.

jsy360, 2020-05-04:
> For MAC users:
> https://apple.stackexchange.com/questions/32834/is-there-an-os-x-terminal-program-that-can-access-serial-ports

Just downloaded Serial. The thing is that it lets me access the Due via USB, but I'm not sure it lets me access Serial1 - suggestions?

fu-hsi, 2020-05-04:
You must connect PMS directly to your computer through USB-RS232.
Arduino Due is not needed at all.

jsy360, 2020-05-05:
Thanks. Nothing has worked so far, so I ordered a new sensor unit. It comes on Thurs. Will let you know if it works. Thanks again for all the assistance @fu-hsi @munzc

closed 2020-05-06 by fu-hsi

jsy360, 2020-05-07:
hi @fu-hsi and @munzc . My new unit arrived and it turns out that it was a faulty unit. I plugged in a new unit and everything works fine. Thanks for all the assistance :) 

Hope you stay safe during this plague, and I wish you both good fortune in the wars to come ;)

[image]

sarita-sudo, 2020-09-08:
Hi! Thank you so much for starting this discussion I'm having very similar issues as you did. Did you end up using the Arduino Due board for the test that worked? And was your code what I have included below?

#include <PMS.h>

PMS pms(Serial1);
PMS::DATA data;

void setup()
{
  Serial.begin(115200);
  Serial1.begin(9600); 
}

void loop()
{
  if (pms.read(data)){

    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);
    Serial.println();
    delay(1000);
  }
}

fu-hsi, 2020-09-09:
You can't use delay() on that way, because you break connection between PMS.
If you want use delay(), use this code:
https://github.com/fu-hsi/PMS/blob/master/examples/Advanced/Advanced.ino

sarita-sudo, 2020-09-10:
Hi, thank you for your response!

I have taken your advice and changed my code to the example without delay as shown here: 

```
#include "PMS.h"

PMS pms(Serial);
PMS::DATA data;

void setup()
{
  Serial.begin(9600);   
  pms.activeMode();  
}

void loop()
{
  if (pms.read(data))
  {
    Serial.print("PM 1.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_1_0);

    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);

    Serial.println();
  }
}
```

My wiring is Tx->D1 and Rx->1 and the PMS is powered and grounded all by the Elegoo Uno. I have two PMS5003 as my final goal will be to display the difference between the two and have tried to switch out sensors (checking for a faulty sensor), plug in the sensor wires after the code is finished uploading, and ran code from both my pc and laptop,  however my output continues to be:
[image]

I should also note that the first time I ran this code, it worked, but no matter what I've tried, haven't been able to get it after that first time. 

I'm really not sure what could be wrong, would you have any other suggestions?

fu-hsi, 2020-09-10:
Above code is also wrong :), You did not set Serial speed.
Please, always use full code example before any modifications.
https://github.com/fu-hsi/PMS/blob/master/examples/Basic/Basic.ino
Why not use simple Nano or ESP? I can see that all problems with PMS comes from Uno users only :).
Unfortunately I don't have it (or maybe I have one) and I only have one PMS in production.
I can't use it for testing with another board now :(
Too much time has passed since the project was created., which was well tested in its time and works for me so far.

sarita-sudo, 2020-09-10:
Okay, I tried that code with no luck. No worries, thank you for your guidance. I will try to use the Due instead, just really curious about why it worked perfectly once on the UNO, will post here if I find a solution.

munzc, 2020-09-10:
> My wiring is Tx->D1 and Rx->1 and the PMS is powered and grounded all by the Elegoo Uno.

Hi, my code that you tried should work with the UNO.  Although I think maybe you have it wired wrong.  You shouldn't have any wires plugged into the arduinos hardware serial ports (Pin 0 & Pin 1).  If you did that, that is probably why you're getting those weird outputs.

The AltSoftSerial library makes pin 8 and pin 9 software serial ports, these are the pins that are connected with the PMS sensor.  **So try this pinout:**
  Sensor           Arduino
Pin 4 (RX)  ->  Pin 9 (TX)
Pin 5 (TX)  ->  Pin 8 (RX)