fu-hsi/PMS #21: Deep sleep problem

Kreeseek, 2020-05-21:
Hi guys,
Should not `pms.sleep();` turn the fan off? I have feeling that in my case sleep does not work well, because I do not see power consumption difference between an active mode and sleep mode. And at this point, I'm wondering if it is only software sleep and fan should work all the time or maybe it should sleep as well but my code is not working properly :)

Thank you@

fu-hsi, 2020-05-21:
Hi.
I guess, your code/connection is wrong.
You can also put the module to sleep manually.
From PMS7003 documentation:

> PIN10 SET Set pin /TTL level@3.3V，high level or
> suspending is normal working status, while
> low level is sleeping mode

I don't know what module and code do you use.
Perhaps commands is not received because of a bad connection.

Kreeseek, 2020-05-21:
So fan indeed should be turned off while sleeping, is that right? :)
Thanks a lot!

fu-hsi, 2020-05-21:
Yes.

closed 2020-05-21 by fu-hsi