fu-hsi/PMS #24: Can't get it to display pm values on esp32

BasvdVelde, 2020-11-09:
Hi,

I hope this is still active, as I would really like to use your code for my PMS5003 on the ESP32 WROOM. When uploading the basic code (after downloading the library) to my esp32 it displays thefollowing:

⸮!⸮H⸮>⸮\x⸮K⸮$⸮

At first I thought it had something to do with the baud rate, but this is set to 9600.

Do you have a clue what might be the issue here?

Thanks,

Bas

mienho, 2020-12-31:
Hello,
I have the similar problem with my PMS7003 on ESP8266. Only display this values all the time: 
[image]

novumsoft-projects, 2021-08-29:
This is common issue of hardware problem in hardware **tty**

cydavidh, 2022-02-20:
Don't use the RX/TX pins (pin 0 and 1 on arduino). This serial port is used for communication between your arduino/esp and computer.
Instead, use the built-in SoftwareSerial library to open another one, like this:

`#include <SoftwareSerial.h>

#include "PMS.h"

SoftwareSerial pmsSerial(D8, D7);  
//SoftwareSerial(rxPin, txPin), I used D8 for RX and D7 for TX (esp8266) and connected TX on PMS5003 to D8

PMS pms(pmsSerial);
PMS::DATA data;

void setup()
{
  Serial.begin(9600);   
  pmsSerial.begin(9600);  
}

void loop()
{
  if (pms.read(data))
  {
    Serial.print("PM 1.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_1_0);

    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);

    Serial.print("PM 10.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_10_0);

    Serial.println();
  }

  // Do other stuff...
}`