fu-hsi/PMS #26: Question about sleep and wakeup

Belphemur, 2022-01-28:
Hello,

I went through the spec sheet of the PMS5003 and I have a couple of question about sleep/wake that I couldn't find in the specsheet neither the lib documentation and I hope you can help me with them.

## Question
1. Does the WakeUp/Sleep works in active mode or only in passive mode ?
2. Is the PIN 3 (SET) needed for that ? Or does the commands goes through the normal serial port ?

fu-hsi, 2022-01-28:
Hi.
1. Both.
2. No and yes. You can sleep device using commands (serial port) or using hardware (pin).

Belphemur, 2022-01-28:
@fu-hsi Thanks for this information. That confirm my assumption, but I wasn't completely sure.

Do you know what is the usage of the PIN 3 (SET) then ? Since, from my understanding, you can do the same through the commands.

fu-hsi, 2022-01-28:
Yes, you can do the same by pin but it's require additional wire and free pin.

Belphemur, 2022-01-28:
Great !

Thanks for all this.
Have a great day.

closed 2022-01-28 by Belphemur