fu-hsi/PMS #29: Cannot get output from Arduino

rudvfaden, 2023-06-11:
Thank you for the library.

I have issue getting output on Arduino r3. I have connected my rxd to pin6 and txd to pin 4 and I use this code.

I was hoping you could guide me in the right direction.

```c
#include "PMS.h"
#include <SoftwareSerial.h>

SoftwareSerial Serial1(4,6);

PMS pms(Serial1);
PMS::DATA data;

void setup()
{
  Serial.begin(9600);   // GPIO1, GPIO3 (TX/RX pin on ESP-12E Development Board)
  Serial1.begin(9600);  // GPIO2 (D4 pin on ESP-12E Development Board)
}

void loop()
{
  if (pms.read(data))
  {
    Serial1.print("PM 1.0 (ug/m3): ");
    Serial1.println(data.PM_AE_UG_1_0);

    Serial1.print("PM 2.5 (ug/m3): ");
    Serial1.println(data.PM_AE_UG_2_5);

    Serial1.print("PM 10.0 (ug/m3): ");
    Serial1.println(data.PM_AE_UG_10_0);

    Serial1.println();
  }

  // Do other stuff...
}
```

Just-ice-E, 2023-07-29:
I have a similar issue, I cannot get my sensor (PMS5003) to interface with my Nano and output to Serial Monitor. I have a DHT sensor interfacing also to verify that the code is working. 

Here is my code:

#include "DHT.h"
#include "PMS.h"
#include "SoftwareSerial.h"

#define DHTPIN 13     // Digital pin connected to the DHT sensor
#define DHTTYPE DHT22   // DHT 22  (AM2302), AM2321

DHT dht(DHTPIN, DHTTYPE);

SoftwareSerial Serial1(3,2); // RX, TX 
PMS pms(Serial1);
PMS::DATA data;

void setup() {
  Serial.begin(9600);
  Serial.println(F("DHTxx test!"));

  dht.begin();

  Serial1.begin(9600);  
  Serial.println("Warming up");
}

void loop() {
  // Wait a few seconds between measurements.
  delay(2000);

  float h = dht.readHumidity();
  // Read temperature as Celsius (the default)

  // Check if any reads failed and exit early (to try again).
  if (isnan(h)) {
    Serial.println(F("Failed to read from DHT sensor!"));
    return;
  }

  Serial.print(F("Humidity: "));
  Serial.println(h);

  if (pms.read(data)) {
    Serial.println("PM reading good");         //This is the real problem area, I can't even get the code to output anything
  }
}

fu-hsi, 2023-07-29:
@Just-ice-E you can't block main loop like these:

> // Wait a few seconds between measurements.
> delay(2000);

Please use examples provided with library.
You can also use passive mode and then you will be able to block with the main loop.